doceme/py-spidev #15: PyPi Package Abandoned?

Gadgetoid, 2015-01-29:
I noticed the package on PyPi hasn't been updated since 2012-12-17. A number of people, particularly in the Raspberry Pi community, depend upon this package for SPI access in Python- and as of Kernel 3.15+ it's going to break ( see recent pull request about this ).

Are you able to either:

a) Accept the pull request and publish this package to PyPi

or

b) Turn over the PyPi package to us, so that we can maintain it better for the benefit of all?

doceme, 2015-01-29:
I have never been involved in the PyPi package. I believe someone else created it after seeing my github repository.  I would reach out to the owners/maintainers of that package directly, if possible.

I apologize for not being able to maintain this project properly, but I don't have the time to do it justice.  If someone wants to take the reigns on this github project, I am happy to update the Readme file on this repository, mention it's been deprecated, and to suggest people use a different fork moving forward.

I will take a look at this pull request today and see if I it works on pre-3.15 kernels.  Assuming it looks okay, I'll pull it in.

Gadgetoid, 2015-01-29:
Thank you, you're awesome.

I've begun the hunt for the PyPi package maintainers, and will reach out to them and see what can be done.

I'll keep you posted!

Gadgetoid, 2015-01-29:
I've had a great response from the original publisher. The project he needed spidev for no longer needs it, so I now have full access - as Pimoroni so it's upkeep is in safe hands! I will thoroughly test these changes and try to reach out to other spidev users and give them a heads up before publishing the updated package.

I'd be willing to fork this package under the Pimoroni GitHub account if you're okay designating that as the "replacement". Although I'm certainly not precious about taking on the responsibility- just as long as the "official" fork is kept up to date.

doceme, 2015-01-29:
Thanks, that sounds good.  If you're willing to help out with testing new pull requests, I can continue maintaining this for now.

Gadgetoid, 2015-01-30:
I'm amassing all the change requests, improvements and general maintenance tweaks into a fork here: https://github.com/gadgetoid/py-spidev

This includes boring stuff like documentation, changelog, etc!

Once complete and thoroughly tested, I'll submit the whole lot as a pull-request and push v3.0 up to PyPi.

doceme, 2015-01-30:
Thanks for working on this!

doceme, 2015-07-29:
Hey Gadgetoid, are you still working on py-spidev?  Can you either make a new release (3.1?) on PyPi that includes the latest fixes from my repo or hand over access to spidev on PyPi to me so I can make the release?  I notice that you don't have issue requests enabled on your fork.  Is that intentional?

Gadgetoid, 2015-07-29:
Ahoy! I'm not actively maintaining it at the moment, and clearly had unintentionally forgot to enable issues on my fork! It's a stupendously important library in our product ecosystem, and for the Raspberry Pi community in general, though so it's clearly something I need to be watching closely!

I'll see what's new, and see about granting you PyPi access so the project doesn't just have one maintainer.

closed 2015-07-29 by doceme

Gadgetoid, 2015-07-29:
I'm happy to push the current state up as 3.1 once I've compiled and tested on a Pi 2. It seems to build fine on an older Pi but I'd like to be sure.

If you want to be involved with the PyPi package, let me know your username and I'll try and add you as a Maintainer. Otherwise I'm happy for you to poke me when it needs deploying!

I'd suggest tagging the GitHub repo as 3.1 when it's ready to go, for posterity.

doceme, 2015-07-29:
My username is also doceme on pypi.python.org (PGP Key ID: 00F17356). I'm happy to let you deploy it this time, but might not hurt to have more than one maintainer just in case.  I finally got a Pi 2 and that's what I used to test this one, but please feel free to test before deploying.

Gadgetoid, 2015-07-29:
Deployed, and added you as a maintainer just in case.

doceme, 2015-07-29:
Excellent, thanks!