doceme/py-spidev #20: Delay does not delay in between blocks

dhhagan, 2015-03-28:
It appears that `delay_usec` delays the chip select deactivation, but nothing else. When sending multiple bytes, there is not delay in between the bytes which is how I interpret the purpose of `delay_usec`. Is that even the correct interpretation? See image below: Sent two bytes with `delay_usec = 9000`

[image]

doceme, 2015-04-01:
Unfortunately, I think delay_usec is only designed to delay between transfers, but not between each byte.  See https://groups.google.com/forum/#!topic/linux-sunxi/61CJ4tFaGMQ.

closed 2015-07-29 by doceme

dany0792, 2019-11-05:
Hi,
I use your project(spidev 3.4) to make a communication between (personal development board) and (ASUS Tinker Board S). I use xfer2 function to read spi data on ASUS Board from my board. So i want to add a delay_usec in between blocks, I found the delay_usec in the spi.xfer2 but it is not working for me. I have tried several ways to add this delay, but I didn't succeed.
Some of my attempts I used are:

spi = spidev.SpiDev()
spi.open(2,0)
spi.mode = 1
#spi.max_speed_hz = 250000
spiResp = spi.xfer2([0, 1, 2, 3], 250000, 1000, 8)

or:
spiResp = spi.xfer2([1])
spiData = range(spiResp[0])
spiResp = spi.xfer2(spiData, 250000, 1000)

Please, help me with that ...
Thank you!