doceme/py-spidev #26: Build failure: conflicting types for ‘fd_set’ et al.

GrantEdwards, 2015-06-29:
I'm trying to install for Python 2.7 with kernel 3.18, and I get
dozens of compiler errors (see below).  So far, Google hasn't found me any
answers (only people reporting the same problem).  Any ideas?

------------------------------------------------------------------------
```
# sudo python setup.py install

running install
running build
running build_ext
building 'spidev' extension
x86_64-pc-linux-gnu-gcc -pthread -fPIC -I/usr/src/linux/include -I/usr/include/python2.7 -c spidev_module.c -o build/temp.linux-x86_64-2.7/spidev_module.o
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:14:26: error: conflicting types for ‘fd_set’
 typedef __kernel_fd_set  fd_set;
                          ^
In file included from /usr/include/sys/types.h:219:0,
                 from /usr/include/stdlib.h:314,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/select.h:75:5: note: previous declaration of ‘fd_set’ was here
   } fd_set;
     ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:15:25: error: conflicting types for ‘dev_t’
 typedef __kernel_dev_t  dev_t;
                         ^
In file included from /usr/include/stdlib.h:314:0,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/types.h:60:17: note: previous declaration of ‘dev_t’ was here
 typedef __dev_t dev_t;
                 ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:19:17: error: conflicting types for ‘nlink_t’
 typedef __u32   nlink_t;
                 ^
In file included from /usr/include/stdlib.h:314:0,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/types.h:75:19: note: previous declaration of ‘nlink_t’ was here
 typedef __nlink_t nlink_t;
                   ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:25:26: error: conflicting types for ‘timer_t’
 typedef __kernel_timer_t timer_t;
                          ^
In file included from /usr/include/sys/types.h:132:0,
                 from /usr/include/stdlib.h:314,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/time.h:103:19: note: previous declaration of ‘timer_t’ was here
 typedef __timer_t timer_t;
                   ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:45:26: error: conflicting types for ‘loff_t’
 typedef __kernel_loff_t  loff_t;
                          ^
In file included from /usr/include/stdlib.h:314:0,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/types.h:44:18: note: previous declaration of ‘loff_t’ was here
 typedef __loff_t loff_t;
                  ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:111:17: error: conflicting types for ‘uint64_t’
 typedef  __u64  uint64_t;
                 ^
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/4.8.4/include/stdint.h:9:0,
                 from /usr/include/inttypes.h:27,
                 from /usr/include/python2.7/pyport.h:9,
                 from /usr/include/python2.7/Python.h:58,
                 from spidev_module.c:20:
/usr/include/stdint.h:55:27: note: previous declaration of ‘uint64_t’ was here
 typedef unsigned long int uint64_t;
                           ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:112:17: error: conflicting types for ‘u_int64_t’
 typedef  __u64  u_int64_t;
                 ^
In file included from /usr/include/stdlib.h:314:0,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/types.h:203:1: note: previous declaration of ‘u_int64_t’ was here
 __u_intN_t (64, __DI__);
 ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:113:17: error: conflicting types for ‘int64_t’
 typedef  __s64  int64_t;
                 ^
In file included from /usr/include/stdlib.h:314:0,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/types.h:197:1: note: previous declaration of ‘int64_t’ was here
 __intN_t (64, __DI__);
 ^
In file included from /usr/include/linux/spi/spidev.h:25:0,
                 from spidev_module.c:25:
/usr/src/linux/include/linux/types.h:134:23: error: conflicting types for ‘blkcnt_t’
 typedef unsigned long blkcnt_t;
                       ^
In file included from /usr/include/stdlib.h:314:0,
                 from /usr/include/python2.7/Python.h:42,
                 from spidev_module.c:20:
/usr/include/sys/types.h:248:22: note: previous declaration of ‘blkcnt_t’ was here
 typedef __blkcnt64_t blkcnt_t;    /* Type to count number of disk blocks.  */
                      ^
spidev_module.c:35:0: warning: "PyLong_Check" redefined [enabled by default]
 #define PyLong_Check(val) PyInt_Check(val)
 ^
In file included from /usr/include/python2.7/Python.h:88:0,
                 from spidev_module.c:20:
/usr/include/python2.7/longobject.h:14:0: note: this is the location of the previous definition
 #define PyLong_Check(op) \
 ^
error: command 'x86_64-pc-linux-gnu-gcc' failed with exit status 1
```

GrantEdwards, 2015-06-29:
In order to get it to build I had to:
1)   Get rid of this line in setup.py:
```
     include_dirs = ["/usr/src/linux/include"],
```
 2) Get rid of this line in spidev_module.c:
 ```
      #define PyLong_Check(val) PyInt_Check(val)
```
Now it builds cleanly and installs.  Don't know if it works yet...

doceme, 2015-07-03:
Can you try with the latest commit?

closed 2015-07-03 by doceme