doceme/py-spidev #28: All xfer2()s on MCP3008 return "TypeError: Argument must be a list of at least one, but not more than 4096 integers"

timwood0, 2015-07-28:
I have followed numerous guides (that all agree) to wire up the MCP3008.  I breadboarded a simple pot sensing circuit.  The spidev/spi_bcm{2708,2835} drivers are installed and I tested MISO/MOSI with a direct test.  But I always get this message.  
Traceback (most recent call last):
  File "adafruit_mcp3008-spi.py", line 56, in <module>
    trim_pot = readadc(potentiometer_adc)
  File "adafruit_mcp3008-spi.py", line 36, in readadc
    r = spi.xfer2(commandout)
TypeError: Argument must ...

The skinny on the affected code (ported from Adafruit bit-banging to use py-spidev):
def readadc(adcnum):
	if ((adcnum > 7) or (adcnum < 0)):
		return -1
	commandout = [ 0, 0, 0 ]
	commandout[0] = 0x1 # start-bit
	commandout[1] = (0x8 | adcnum) << 4 # single-ended bit | A-channel#
	commandout[2] = 0   # we only need to send 5 bits here

	# Take in result as array of three bytes. 
	# Return the two lowest bits of the 2nd byte and
	# all of the third byte; use default speed, bits/word, delay
	r = spi.xfer2(commandout) <-- ERROR

	# read in 10 ADC bits
	adcout = ((r[1] & 3) << 8) | r[2]
	return adcout

BTW spidev_module.c confusingly uses the same error message for several different error cases.

Anyone see anything obviously wrong here?  I tried running with tomstokes/python-spi library; the test senses bursts of noise values on the A/D output with that stack.

timwood0, 2015-07-28:
PS As tdicola reports, development of py-spidev is supposed to have moved to Gadgetoid/py-spidev, but that repo does not allow issue reports and all changes since have happened here.

timwood0, 2015-07-28:
PPS The bitbang code from Adafruit works as directed on my RPi+breadboard.

doceme, 2015-07-29:
I'm not the best and noticing the nuances between Python 2 and Python 3. I added a proper fix to support int vs. long type checking.  Can you test with the latest commit on the master branch?

doceme, 2015-07-29:
BTW, I haven't received a firm commitment from Gadgetoid about taking this thing over, so I'm still supporting it as I have time.

timwood0, 2015-07-29:
Now that's service! :)

Your patch for the Int/Long check diffs between Python 2 & 3 seems to have fixed it.

I'll chip in the change to make the error messages more specific, and send a pull req., if you'd like.

Thank you, Doceme!

doceme, 2015-07-29:
Great! Glad to hear it. I always appreciate pull requests.

closed 2015-07-29 by doceme