doceme/py-spidev #31: IOError: [Errno 24] Too many open files

codylallen, 2015-10-06:
When implemented for a large stream of data to send, the script crashes with "IOError: [Errno 24] Too many open files". 

Method of implementation is:
def SendSong(bytes):
   speed = 2500000
    print("Sending packets . . . ")
    while(len(bytes) > 0):
        if(len(bytes) > 4096):
            SPI_sendPackets(bytes[:4096], speed)
            bytes_read = bytes[4097:]
    else:
        SPI_sendPackets(bytes_read, speed)
        bytes_read = list()

def SPI_sendPackets(listOfPackets, speed):
    spi = spidev.SpiDev()
    spi.open(0,0)
    spi.max_speed_hz = speed

    spi.xfer2(listOfPackets)

doceme, 2015-10-06:
The problem is you are calling open from within SPI_sendPackets and not closing it.  Doing this many times over and over gives you the error message you are seeing, "Too many open files."

You shouldn't be opening and closing spidev for every transfer.  Just create a spi object somewhere toward the beginning of your code and only call open on it once.  Then just pass the spi object to SPI_sendPackets and you should be fine.

It looks like you have one other error in your code.  bytes[:4096] will give you bytes 0-4095 (the first 4096 bytes), but bytes[4097:] will skip byte 4096 and give you bytes 4097 to the end of the list.  I'm not sure you meant to skip that byte.

closed 2015-10-06 by doceme

codylallen, 2015-10-06:
Hello,

Thank you for your fast response and assistance. I am embarrassed I made such an amateur mistake! 

Have a great day. 

Cheers,
Cody

////////////////////////////////////////////
Cody L. Allen
Purdue University
College of Engineering
ECESS Vice President
Mobile: 765.299.2447
////////////////////////////////////////////

doceme, 2015-10-06:
No problem. Happy to help. I'm sure I've done worse many times. That's why I love open source.

rajdpandey, 2019-04-08:
> The problem is you are calling open from within SPI_sendPackets and not closing it. Doing this many times over and over gives you the error message you are seeing, "Too many open files."
>
> You [...]

Sir, I am Getting Same Issue But With MFRC522 Library Which Uses Spidev. And I am Not been able to solve This.