doceme/py-spidev #32: "with" statement support

ArmlessJohn404, 2015-11-22:
I've been searching for some documentation to implement `with` statement support for `SpiDev` and couldn't find much. 
Is there interest in adding this feature?  If you point me some directions I can create a PR for that.

doceme, 2015-11-25:
I think that's a good idea, but not something I'm planning to work on anytime soon.  Feel free to create a PR for it if you get it working.

closed 2015-11-25 by doceme

ArmlessJohn404, 2015-11-25:
I'm not quite comfortable writing `python` extensions in `C`, I'll take another look. Do you know any resources I can read about the `with` statement?

doceme, 2015-11-25:
I don't expect you to implement it, but just wanted to warn you that I probably won't have time to add support for this soon.  It looks like two new functions would need to be defined called "__enter__" and "__exit__".  Have a look at this discussion: http://www.gossamer-threads.com/lists/python/python/949495 and this code as an example: https://github.com/jolicloud/python-apt/blob/master/python/lock.cc

Something similar would need to be added.

doceme, 2015-11-25:
Looks like github strips out the underscores.  There are supposed to be two underscore characters before and after the __enter__ and __exit__ above.

ArmlessJohn404, 2015-11-25:
ok! thanks!