doceme/py-spidev #33: package extension

ArmlessJohn404, 2015-11-22:
I'm implementing a [library](https://github.com/ArmlessJohn404/RPi_mcp3008) for the MCP3008 A/D converter chip. 
It just implements the example communication protocol in the [datasheet](https://www.adafruit.com/datasheets/MCP3008.pdf) inheriting `SpiDev`.
Is there interest in creating an `extension` folder for the package to add support for devices like these?

doceme, 2015-11-25:
There are an endless number of chips that this library could potentially communicate with.  I think that code would be best served in another repository.  Feel free to create one and reference this one, or fork this one if you don't agree.

closed 2015-11-25 by doceme

ArmlessJohn404, 2015-11-25:
Allright! thanks