doceme/py-spidev #35: xfer does not de-assert CSbetween bytes

EricPobot, 2015-12-21:
xfer seems to behave exactly as xfer2, and does not de-assert CS between the bytes of the data to be transmitted. Checked with logic analyzer.

Context: Raspi2, Raspbian, Python 2.7.9

semiversus, 2017-06-07:
This is my understanding what xfer should do. Set CS to active, transfer given bytes and deassert at the end.

semiversus, 2017-06-07:
The difference between xfer and xfer2 is the handling of CS at the end of the transmission: deassert it or leave it active.

Gadgetoid, 2020-05-29:
I know I'm late to the party, but I've been digging through the code lately. I'm pretty sure there is no difference between `xfer` and `xfer2` unless the library is compiled with `-DSPIDEV_SINGLE`. Otherwise these two methods are functionally identical.