doceme/py-spidev #36: Broken on latest release/firmware

malcb, 2016-01-02:
I tried to get this and the Gadgetoid fork to work but couldn't.  I then tried the instructions here

https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting

and built the c spi test program. That works if I link MISI MOSI and gives zeros if I break the link so nothing wrong with my configuration.  It seems pretty certain this code is broken on latest release/firmware.  The driver has changed to spi_bcm2835 if that is relevant.

obtitus, 2016-05-11:
Seems to work fine on a new raspberry 3 and a old raspberry 1. Could you provide a MWE?

fschrempf, 2025-09-25:
Closing this as outdated and missing information. Please open a new issue if problem still exists.