doceme/py-spidev #40: silent failure in python when using xfer() and xfer2()

prygaard, 2016-03-04:
I recently tried running this code:
``import sys
import spidev
spi_bus = spidev.SpiDev()
exit_code = spi_bus.open(0, 0)
print('getting the adc output ')
adc_output = spi_bus.xfer2(1, 240, 0 )
print('adc read complete')``

The code would get to the xfer2() and the just stop and return to the prompt with no error or warning.

The bug in the above code is that there should have been brackets around the list passed to xfer2() as shown below.

``import sys
import spidev
spi_bus = spidev.SpiDev()
exit_code = spi_bus.open(0, 0)
print('getting the adc output ')
adc_output = spi_bus.xfer2([1, 240, 0] )
print('adc read complete')``

Ideally SpiDev.xfer2() would have generated some type of error or warning rather than have the whole program just stop without an error message.   

Note:  I am a newbee so I do not even know if it is possible to do what I am suggesting.   However, if you can, it would sure save the next person to hit the problem a lot of effort.

prygaard, 2016-03-04:
Hmmm.   My code did not format well in my post.   Here is another try:

I recently tried running this incorrect python 3 code with spidev:

     import sys
     import spidev
     spi_bus = spidev.SpiDev()
     exit_code = spi_bus.open(0, 0)
     print('getting the adc output ')
     adc_output = spi_bus.xfer2(1, 240, 0 )
     print('adc read complete')  

The code would get to the xfer2() and then just stop and return to the prompt with no error or warning. 
(The last print statement would never execute)

The bug in the above code is that there should have been brackets around the list passed to xfer2() as shown below.

     import sys
     import spidev
     spi_bus = spidev.SpiDev()
     exit_code = spi_bus.open(0, 0)
     print('getting the adc output ')
     adc_output = spi_bus.xfer2([1, 240, 0] )
     print('adc read complete') 

Ideally SpiDev.xfer2() would have generated some type of error or warning rather than have the whole program just stop and return to the prompt without an error message. 

Note: I am a beginner so I do not even know if it is possible to do what I am suggesting. However, if you can, it would sure save the next person to hit the problem a lot of effort.

semininja, 2016-03-07:
I agree, I would have expected a TypeError either for too many arguments or arguments not being lists. Just out of curiosity, does it work with tuples?

prygaard, 2016-03-07:
I won't be able to get to it for a couple days, but my guess is it won't.   I had tried this: 
xfer2(0) and it did not work.
The readme documentation says the parms are:

`xfer2(list of values[, speed_hz, delay_usec, bits_per_word])`

I *think* the first parameter must be a list with the '[ ]' brackets.  When I get a chance I will play with it some more.

I tried looking at the wrapper code

prygaard, 2016-03-07:
It seems to me that no code should abort and not provide some kind of error or warning.

I tried looking at the C wrapper code but quickly got lost in the magic of python calling c code.   Someone more experienced than I will be needed to figure this out.

doceme, 2016-03-07:
I agree that it should not fail silently. I'll try to take a look at it soon.

doceme, 2016-03-07:
Please check the latest code and confirm that TypeError is raised when specifying a non-list type.

prygaard, 2016-03-07:
That worked like a charm!  

this statement works fine:
`adc_output = self.spi_bus.xfer2([1, 240, 0])`

This statement fails gracefully with an error message
`adc_output = self.spi_bus.xfer2(1, 240, 0)`

The error message is:
`TypeError: must be list, not int`

Thank you for fixing this!!   I am sure it will help the next guy to hit the problem a *lot*.

I am new to all of this.   Is there something that needs to be done to get the fix into other branches of spidev such as (https://pypi.python.org/pypi/spidev/3.0)

closed 2016-03-07 by prygaard

doceme, 2016-03-07:
No problem, thanks for pointing it out!  I just need to test a few more things on my RPi and Beaglebone to confirm that the latest commits haven't broken anything.  Then, I'll make a new PyPi release that includes this fix.

prygaard, 2016-03-07:
Thanks!! Paul T. Rygaard 
 Software Development Executive 
408-438-3102 
 prygaard@yahoo.com      www.linkedin.com/in/rygaard

doceme, 2016-03-08:
Done: https://pypi.python.org/pypi/spidev/3.2

prygaard, 2016-03-08:
Wow!    That was fast.   Thanks!!

doceme, 2016-04-26:
Looks like I should have used PySequence_* functions to handle this. I created a pull request on the fix-list branch to address this. Would anyone care to test it out before I merge?