doceme/py-spidev #42: How to sets values to differents CS?

sophiekovalevsky, 2016-03-21:
Thanks for this amazing library. 
I was wondering how can I set values to differents chip select lines due the fact that I'm not sure which line is set when I'm using the property cshigh. 
Thanks in advance.

Gadgetoid, 2016-03-21:
I've been grappling with low-level SPI recently, trying to figure out how to not assert a chip-select line at all so I can control 4 devices instead of two. So hopefully I'm not going to spout complete nonsense.

The chip select lines correspond to the SPI device you pick with `spi.open()`. The first argument being the SPI bus number and the second being either 0 for CS0 ( BCM 8 ) or 1 for CS1 ( BCM 7 ) : http://pinout.xyz/pinout/spi

If you want to use any pin, you can just assert it manually before an `xfer`, while it's super handy to have the SPI library take care of chip-select for you ( and probably ever so slightly faster ), it's certainly not necessary.

sophiekovalevsky, 2016-03-22:
Thanks for the quick response. 
So, if I have to instance of the spi library, which mean that I'm using the two ports that my embedded offer, how do I know which port affect the cshigh method?

Thanks a lot.

obtitus, 2016-05-11:
hi sophiekovalevsky, the following example demonstrate how you can set channel 0 to cshigh=True and channel 1 to cshigh=False. I am not sure this is the only way to achieve this, but it seems to work.

```python
import spidev
spi0 = spidev.SpiDev()
spi0.open(0, 0)
spi0.cshigh = True

spi1 = spidev.SpiDev()
spi1.open(0, 1)
spi1.cshigh = False
```
You can now use say `spi0.xfer` for channel 0 and `spi1.xfer` for channel 1.

sophiekovalevsky, 2016-05-17:
Thanks a lot.

closed 2020-03-03 by sophiekovalevsky