doceme/py-spidev #48: max_speed_hz does not work

tnedev, 2016-08-22:
Hi, 

I'm using spidev to communicate with LCD displays. 
I found that setting max_speed_hz has no effect on the actual SPI speed which continues to run at 64Mhz. 

Kernel: 4.1.21-v7+

Gadgetoid, 2016-08-22:
Could you post a minimal code example to reproduce this? I've just tested this with the following code and verified via my scope that it's taken effect:

```
import time
import sys
from spidev import SpiDev

dev = SpiDev(0,0)
dev.max_speed_hz = 10

try:
    while True:
        dev.writebytes([1,2,4,8,16,32,64,128])
        time.sleep(0.001)
except KeyboardInterrupt:
    sys.exit(0)
```

I believe the speeds, in hz, snap to the nearest available speed due to the way the hardware works. It could be that you're trying to make a change too small, so it's just snapping to 64Mhz ( should be 62.5Mhz )

A table of speeds is listed here: https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md

tnedev, 2016-08-22:
I'm using the SPIDev through Adafruit Python ILI9341 and GPIO Libraries

Adaftuit GPIO Python Library: https://github.com/adafruit/Adafruit_Python_GPIO
Adafruit Python ILI9341: Library: https://github.com/adafruit/Adafruit_Python_ILI9341

I'm running extended version of this example code:https://github.com/adafruit/Adafruit_Python_ILI9341/blob/master/examples/image.py

Basically, Adafruit Python GPIO SPI provides a  wrapper for the SPIDev
Changing the clock speed of the Python SPIDev wrapper does not change the clock speed which I confirmed with logic analyzer. 
My guess was that the problem comes from the SPIDev library. 
I will test with your example code and confirm if it works with me. 
 

From what

tnedev, 2016-08-22:
Hi, 

I confirm that it works well through the original SPIDev. It seems that the Adafruit wrapper messed it up somehow. 

Case closed.

closed 2016-08-22 by tnedev