doceme/py-spidev #54: install error

J-Arji, 2017-02-06:
hi
when that i install setup.py 
`sudo python setup.py install
`
error : 
r```
unning install
running build
running build_ext
building 'spidev' extension
creating build
creating build/temp.linux-armv7l-2.7
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c spidev_module.c -o build/temp.linux-armv7l-2.7/spidev_module.o
spidev_module.c:20:20: fatal error: Python.h: No such file or directory
 #include <Python.h>
                    ^
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

rm-hull, 2017-02-06:
I think you most likely need the `python-dev` package installing, eg

    $ sudo apt-get install python-dev

jamesoflol, 2017-09-13:
This worked for me. Thanks

closed 2017-10-10 by J-Arji