doceme/py-spidev #55: xfer2 glitch on cs.

ead0601, 2017-02-08:
Hi,

I am using a Pi 3 version 8 (jessie), and py-spidev (HEAD) (Merge: 511db0d 11adf8c)

When I execute:

resp = spi.xfer2([0xD8, 0x00, 0x00, 0x00])

I see a glitch on CS during the transfer.

Thanks,
Edward

ead0601, 2017-02-11:
I wanted to add that the glitch occurs near the begining of the transfer, and for the rest of the burst of 4 bytes, the glitch is not there.

ead0601, 2017-02-17:
I added the following to /boot/config.txt

core_freq=250

This disables the auto frequency throttling, and fixes the core frequency. 
   1) I no longer see the glitch.
   2) However, very intermittently, I receive the incorrect data. I have slowed down the transfer and ensured proper grounds, and a problem still occurs. The SI looks good on a scope.

Given the improved performance with the "core_freq" setting, I can implement a retry work around. Luckily the first couple of MISO bits are fixed, so I will know when to retry.

Something, is still not Kosher.

Edward

doceme, 2017-02-23:
It sounds like this is a Pi 3 problem or kernel driver issue and not related to spidev.

closed 2017-02-23 by doceme

ead0601, 2017-02-23:
The problem was fixed with the lastest pull request:

https://github.com/doceme/py-spidev/pull/53#pullrequestreview-23496026