doceme/py-spidev #59: Can I set MOSI default HIGH, and set CS low separately for ~1ms?

BOBBEAR82, 2017-07-06:
Hello,
I am doing a project using RPi to talk with Linear LTC6804 (battery monitoring chip) in SPI. I am using Python with spidev module to control SPI communication. However, it looks like the default MOSI pin signal is LOW, while the LTC6804 requests MOSI default HIGH.

See below in good communication the LTC6804 requests MOSI default HIGH. Also before sending request on MOSI, the CS needs to be set LOW for 0.3ms for waking up the chip.
[image]

See below in bad communication with LTC6804 using functions from spidev with MOSI default LOW. I cannot find how to set MOSI pin default HIGH with spidev.
[image]

So see someone can help out:
1. how can I set MOSI pin default HIGH when using spidev in Python? Or some other ways to achieve that?
2. how can I set CS pin LOW for ~1ms when still using spidev in Python? 
I want to get exactly the same output from RPi as the first attached screenshot.

Thank you!

SofiaRafael, 2019-06-13:
Hi, 
Did you arrive at any conclusion about this?

Thank you very much,
Sofia

Troyhy, 2019-11-25:
I'm having similar issues. Any solution?

semininja, 2019-11-27:
You may be able to get the same effect as MOSI default HIGH (also called "active low") by inverting the data you're sending (i.e. flipping each bit before transmission, e.g. converting 0b00110011 to 0b11001100). You can do this easily; if `foo` is a byte of data, instead of `spi.xfer(foo)` you'd `spi.xfer(~foo)`.

semininja, 2019-11-27:
As far as setting the CS pin to wake up the chip, will it work if you just send a byte (or a few bytes) of non-data (e.g. 0x00 or 0xFF etc.) first? My cursory inspection of the datasheet implies that for the LTC6804, that should work, although I haven't taken the time to fully understand the function of the device.

Troyhy, 2019-11-27:
I believe the OP described the situation well.
I have Arduino witch is generating same SPI command, but data line rests HIGH, but is not inverted. CS will wake the device. 
Raspberry sends same command, but  MOSI is resting in LOW state and cannot get chip to wake up.

Problem here is that this is not direct SPI connection, but trough LTC6820 isoSPI bus converter. 
RaspBerry <-- SPI -->  LTC6820  <-- isoSPI --> LTC6804
[image]

Next step is to add OR gate in MOSI line that another GPIO line can force MOSI up during WAKEUP sequence

Troyhy, 2019-11-28:
I can confirm that I got this working with help of additional IO line witch will force MOSI line rest up. Simple solution with 2 shotky diodes was enough here.