doceme/py-spidev #72: strange delay in transmitting after 60-70 bytes of data

melazarus, 2018-04-08:
I'm using (abusing) SPI to send data to a WS2812 LED string.

The code (got it from: https://github.com/joosteto/ws2812-spi):
```
def write2812(spi, data):
    tx=[0x00]
    for rgb in data:
        for byte in rgb:
            for ibit in range(3,-1,-1):
                tx.append(((byte>>(2*ibit+1))&1)*0x60 +
                          ((byte>>(2*ibit+0))&1)*0x06 +
                          0x88)
    spi.xfer2(tx,3200000)
```

Since the ws2812 only uses one wire for data & clock timing is crucial.

This works perfectly for 5 or fewer pixels, above that the behaviour becomes erratic.
So I connected a scope to the SPI (MOSI) Pin and discovered that after about 60-70 bytes the SPI stops for a brief time (about 5µs). 
[https://ibb.co/mD9vZx]

I don't think this is a buffer issue. 

Could this be an issue with the hardware SPI? or maybe the py-spidev?

Oh, I'm using a H2+ CPU on a orange pi zero

pietrodn, 2018-04-18:
Here's a possible solution: https://github.com/joosteto/ws2812-spi/issues/6#issuecomment-382293838

fschrempf, 2025-09-25:
This seems to be outdated and likely unrelated to `py-spidev`.

closed 2025-09-25 by fschrempf