doceme/py-spidev #84: Bad default value for SPI clock speed (max_speed_hz) messing up signalling

mkreider, 2019-02-27:
The default value for the clock speed causes some severe issues with the signalling on my system. I could verify this on the logic analyser and fix it by setting the speed to a sensible value.

Version info for my raspberry pi zero W:
```
pi@raspberrypi:~pip freeze | grep spidev
spidev==3.4
pi@raspberrypi:~ $ uname --all
Linux raspberrypi 4.14.50+ #1122 Tue Jun 19 12:21:21 BST 2018 armv6l GNU/Linux
```

Test script:
```
import spidev
import time
spi = spidev.SpiDev()
spi.open(0,0)
print "Initial speed setting %s Hz" % spi.max_speed_hz
#spi.max_speed_hz = 500000
#print "Corrected speed setting %s Hz" % spi.max_speed_hz

# use chip select CE0, channel 0

while True:
    try:
        response = spi.xfer2([1, 0, 0, 0])
        value = response[2] * 256 + response[3]
        print value
        time.sleep(0.3)
    except KeyboardInterrupt:
        spi.close()
```

When I try this without setting the speed explicitly, I get the following result

[image]

and as output
```
pi@raspberrypi:~/python-spi $ python testspi.py
Initial speed setting 125000000 Hz
...
```
Trying 125MHz is a very odd default value for the de facto standard library for a raspberry pi, who has to struggle for anything more than 500kHz.
If I set it explicitly set spi.max_speed_hz to 500kHz by uncommenting line 6-7, I get a correct clock and data on the logic analyser

[image]

The default should be changed to an 'always works' value such as 100kHz or similar, otherwise the 'Usage' example from the readme doesn't work and that's needlessly frustrating to newbies.

semininja, 2019-11-27:
FWIW, according to the Raspberry Pi Foundation's documentation, RasPi devices can handle up to 125 MHz. Do you have a way to verify that the problem is the hardware and not something about your testing setup (e.g. oscilloscope)?

electror7, 2020-03-10:
I faced this same issue. I was interfacing Raspberry Pi with a 5V device using a bidirectional level converter (TXB0108), and was having issues with talking SPI to the 5V device. The hardware setup works with version 3.0 of the py-spidev library, but wouldn't work with version 3.3. Version 3.0 has a default speed of 500kHz. I also found out that the level converter has a max speed of 100MHz at the configuration I had. 
Setting the speed explicitly to a lower speed worked. Having a lower safer value for the default speed would be recommended, but what I learnt from this is to always set the speed explicitly.

Gadgetoid, 2020-03-10:
The library doesn't contain a default value for `max_speed_hz`, but rather requests it from the kernel with an IOCTL when the device is opened:

https://github.com/doceme/py-spidev/blob/f543ca1a8cdc982e6fdfa5402ca9316f8f84bc7a/spidev_module.c#L1317-L1321

I would venture that the solution for this problem is to *always* specify a `max_speed_hz` because the library doesn't appear to make any guarantees about what this might otherwise be.

I don't buy that 3.0 has a default speed whereas 3.3 does not- neither the code for reading or writing the max speed value has changed in 8 years: https://github.com/doceme/py-spidev/commit/acde79b6ef485f7ff32daf391aef285af5091def

electror7, 2020-03-10:
Thank you for your reply. Good to know that `max_speed_hz` is requested from the kernel.

Regarding the version of the library, I should have mentioned that v3.0 was on Raspbian jessie and this problem I encountered was on Raspbian buster with v3.3.

MetaStasisa, 2021-01-02:
FYI it is stated by the Raspberry Pi org that anything above 50Mhz will be iffy to get working
[image]

While I believe the max speed is as stated 125Mhz. 

I'm not sure if spidev is exclusively used by

closed 2025-09-25 by fschrempf