doceme/py-spidev #85: Connection timed out

me21, 2019-03-29:
I occasionally get `IOError: [Errno 110] Connection timed out`.
SPI initialization and usage code:

```
    spi = spidev.SpiDev()
    spi.open(0, 0)
    spi.max_speed_hz = 500000
    spi.mode = 0b00
    ...
    spi.xfer2(msg_to_mb_cobs)
```

What else should I provide to help you troubleshoot the error?

me21, 2019-03-29:
Update: this happens more likely if I set lower `max_speed_hz`.

Gadgetoid, 2023-11-03:
It's probably unrelated to your specific use case, but I raised a bug for my Pi 5 issue here - https://github.com/raspberrypi/linux/issues/5696

The C code may help future users - who end up here like I did - figure out what's going wrong by providing a nice isolated test case.

biemster, 2024-01-18:
The fix provided in the above issue does not work on my Pi 4. Also, in my case the threshold is 96 bytes, xfer2-ing 95 bytes works fine.

Gadgetoid, 2024-01-18:
See my issue and try the test code with 95/96 bytes. If it still fails then that's a Raspberry Pi issue that can't be fixed here.

Though I see you mention "occasionally," which doesn't really gel with what I ran into- but might still be related to a buffer filling up somewhere.

biemster, 2024-01-18:
Your test code throws the same error, and I agree it's a Raspberry Pi issue. I just wanted to comment that this is still seen currently in case more people run into this, I'll try to get help in the proper channels.

fschrempf, 2025-09-25:
This sounds like a kernel issue. Please open a new issue if the problem persists and you think otherwise.