doceme/py-spidev #86: Error : " Too many Open Files " with MFRC522

rajdpandey, 2019-04-08:
I am Using MFRC522 lib File that uses spi-dev. I am Running it on Raspberry Pi 3b+. and OS is Picroft. I am Continuously Scanning RFID card, to when card is detected it would glow green Led and When is removed from the top of Reader it would glow red. But after of 40-45 sec of working correctly it is showing Error : " Too many Files Open". I am Not able to solve it. 

Anyone Knows How to solve it Please Do   Tell.  @ doceme  @ Stephen Caudle

semininja, 2019-11-21:
Your problem is certainly within your specific source code and implementation, because spidev doesn't open any files. If you link to the repo for your project, we might be able to offer some advice, but this is not related to spidev.

rajdpandey, 2019-11-22:
@semininja  Thanks for your response but it is too late I have solved that thing at that time only almost 6months ago. I forgot even that I had an issue opened here.

closed 2019-11-22 by rajdpandey

silver-cpu, 2023-11-16:
How did you fix the issue?