doceme/py-spidev #87: FileNotFoundError: [Errno 2] No such file or directory

Tarasila, 2019-06-21:
Do I have this error because of the incorrect path specified in spidev_module.c. The default path to look for devices is dev/spidevX.Y, whereas my appear at /sys/bus/spi/devices/spiX.Y.

Wuppdich, 2019-12-13:
i came across the same error because i forgot to enable SPI in the Raspberry Pi settings

PizzaRules668, 2020-01-16:
What version of python are you using?

[Spidev is for python 3.4](https://pypi.org/project/spidev/#history)

tim-seoss, 2023-03-09:
/sys/bus/spi/devices/spiX.Y is part of sysfs, and indicates that one or more SPI bus master devices are present, and have SPI controller drivers loaded and bound to them, with the corresponding chip-select lines configured (bus `X` and chipselect `Y`s in your example).

If you don't have spidev devices, then this is because the spidev driver is not bound to any SPI peripheral devices.  It used to be possible to do this using devicetree rules, or using dynamic binding (e.g. using udev rules).  More recent kernels have deprecated the devicetree mechanisms, so udev rules should be used.

closed 2025-09-25 by fschrempf