doceme/py-spidev #108: "error return without exception set" then try to set cshigh to False.

oosavu, 2020-08-10:
Raspberry PI Zero W.

Works good on Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 

Fails on Raspberry Pi OS (32-bit) Lite May 2020 (Release date:2020-05-27)

`
import spidev
spi = spidev.SpiDev()
spi.open(0, 1)
spi.cshigh = False
`
[image]

JonathonReinhart, 2020-09-08:
I think the problem is here: https://github.com/doceme/py-spidev/blob/a5d82b88bd1c95c9f9ec46e9f7afe08c3323e625/spidev_module.c#L904-L906

`__spidev_set_mode` returns `-1` without calling `PyErr_Set*`, which is then returned by `SpiDev_set_cshigh`.

Running this test code:
```python
#!/usr/bin/env python3
import spidev

spi = spidev.SpiDev()
spi.open(0, 0)

spi.cshigh = False
```

I see only these relevant syscalls in `strace`:

```strace
openat(AT_FDCWD, "/dev/spidev0.0", O_RDWR|O_LARGEFILE) = 3
ioctl(3, SPI_IOC_RD_MODE, 0xbea7c23f)   = 0
ioctl(3, SPI_IOC_RD_BITS_PER_WORD, 0xbea7c23f) = 0
ioctl(3, SPI_IOC_RD_MAX_SPEED_HZ, 0xbea7c248) = 0
ioctl(3, SPI_IOC_WR_MODE, 0xbea7d1d7)   = 0
ioctl(3, SPI_IOC_RD_MODE, 0xbea7d1db)   = 0
```

So because none of the `ioctl` calls fail, I have to assume that the attempt to set the `SPI_CS_HIGH` bit was refused by the kernel driver. So `test`, the readback value, didn't match `mode`, the value attempted to be set.

jfernandz, 2020-12-21:
`cshigh` in fact is `True` by default because if you `print(device.cshigh)` you'll get a True, however I'm having a low-active behavior:

[image]

Why is this?

PR #109 (2022-11-03): Set IOError in __spidev_set_mode if readback differs
Apparently on Raspberry Pi (model A) with `spi_bcm2835` (5.4.51+),  `CS_HIGH` (`0x04`)  is always set.

Consider this test code:
```python
#!/usr/bin/env python3
import spidev

spi = spidev.SpiDev()
spi.open(0, 0)

spi.cshigh = False
```

Previously, it would crash with the following trace:
```
Traceback (most recent call last):
  File "./test.py", line 7, in <module>
    spi.cshigh = False
SystemError: error return without exception set
```

It now raises an `IOError` (`OSError`) which explains exactly what happened:
```
Traceback (most rec [... cut]
commit d8725c4e (2022-11-03): Set IOError in __spidev_set_mode if readback differs

This fixes #108
closed 2022-11-03 by doceme