doceme/py-spidev #110: Configure sampling rate in adxl355 accelerator using spidev

Phuntsho1984, 2020-10-16:
I am using mems adxl355 sensor using adxl355.py code for getting data in raspberry pi using spidev module. I am confused how do i configure the sampling rate to 100 samples per second or 100hz. I am wondering how does max_speed_hz in spidev affect the sampling rate.

noeldum, 2020-10-19:
Also interested in the answer to this.
As far as I understand the sampling rate of the sensor might not be something controlled by the spi library. We might have to explicitly program it (like a timely controlled loop or similar). max_speed_hz is only for the clock of the spi. Might not relate directly to how often we query the slave.

delay_usec for the xfer function has caught my attention however.

> xfer(list of values[, speed_hz, delay_usec, bits_per_word])
> Performs an SPI transaction. Chip-select should be released and reactivated between blocks. Delay specifies the delay in usec between blocks.

could this be used as a mean to control sampling rate of a sensor?

Gadgetoid, 2020-10-19:
As @noeldum suggests, `spi_speed_hz` would only affect the data rate *from* the ADXL355 sensor, and not the update rate from the sensor.

This question isn't relevant to the underlying SPI library, but rather something you should be looking for in the Register Definitions/Register Map section of the ADXL355 datasheet - https://www.analog.com/media/en/technical-documentation/data-sheets/adxl354_355.pdf

Or asking of whoever authored the mentioned `adxl355.py`.

closed 2022-11-03 by doceme