doceme/py-spidev #111: Reading from SPI 0.0 is always zero

manushawijekoon, 2020-11-08:
I am trying to read a temperature value from MAX6675 but the value it reads is always zero. I have checked the signals using an oscilloscope which shows non-zero values produced by MAX chip according to the protocol. The raspberry pi generates correct signals for both CS line and CLK. Please see below my full program. 

I am using Raspbian buster.

`import spidev
import time

spi = spidev.SpiDev()
spi.open(0, 0)
spi.max_speed_hz = 50000
spi.mode = 0
spi.lsbfirst = False
spi.threewire = False

def read16(spi):
    # Read 16 bits from the SPI bus.
    #dumm = [0x00, 0x00]
    raw = spi.readbytes(2)
    #raw = spi.xfer2(dumm)
    if raw is None or len(raw) != 2:
            raise RuntimeError('Did not read expected number of bytes from device!')
    print(str(raw[0]) + ' ' + str(raw[1]))    
    value = raw[0] << 8 | raw[1]
    print('Raw value: 0x{0:08X}'.format(value & 0xFFFFFFFF))
    return value

def readTempC(spi):
    """Return the thermocouple temperature value in degrees celsius."""
    v = read16(spi)
    # Check for error reading value.
    if v & 0x4:
            return float('NaN')
    # Check if signed bit is set.
    if v & 0x80000000:
            # Negative value, take 2's compliment. Compute this with subtraction
            # because python is a little odd about handling signed/unsigned.
            v >>= 3 # only need the 12 MSB
            v -= 4096
    else:
            # Positive value, just shift the bits to get the value.
            v >>= 3 # only need the 12 MSB
    # Scale by 0.25 degrees C per bit and return value.
    return v * 0.25

while(True):
    print(str(readTempC(spi)))
    time.sleep(1.0)
`

phantom1299, 2021-01-12:
@manushawijekoon Did you find the cause of this? Experiencing the same problem.

manushawijekoon, 2021-01-12:
Yes. Totally unrelated to software..
The wire I used for clock was disconnected internally.

phantom1299, 2021-01-12:
Oh, ok thanks.

closed 2022-11-03 by doceme