doceme/py-spidev #112: CS line double pulsing

TonyHSmith, 2020-11-12:
I am running Spidev on a Pi 3B+ (Linux raspberrypi 5.4.51-v7+ #1333) and I have noticed that there appears to be 'double pulsing' on the Chip Select (CS) line.

For example, if I run the following simple program to transfer three packets of data:

`import spidev

spi = spidev.SpiDev()
spi.open(0,0)
spi.mode = 0
spi.max_speed_hz = 10000

spi.xfer([0x08])
spi.xfer([0x88])
spi.xfer([0xAA])`

I get the following waveforms:

[image]

Note how the CS line double pulses at the end of each packet.

If I repeat the transfer with WiringPi using the following program:

`import wiringpi as wiringpi
   
wiringpi.wiringPiSetupGpio()
wiringpi.wiringPiSPISetup(0, 10000)

wiringpi.wiringPiSPIDataRW(0, bytes([0x08]))
wiringpi.wiringPiSPIDataRW(0, bytes([0x88]))
wiringpi.wiringPiSPIDataRW(0, bytes([0xAA]))`

I get the following waveforms:

[image]

Note how the CS line exhibits normal behaviour, i.e., goes high then low just once at the end of each packet. In the SpiDev case the data packets are transferred correctly as can be seen from the bus decode at the bottom of the screen and it has not caused any problems when sending commands to my target device. However, I was wondering if there is a reason for the double pulsing and whether it might be a good idea to 'fix it' so that CS goes high once, and only once, at the end of each packet.

Many thanks,
Tony Smith.

semininja, 2020-12-10:
Does this behavior occur when using `writebytes()` instead of `xfer()`?

TonyHSmith, 2020-12-11:
Hi Semininja,

If I run the following:

[image]

I get:

[image]

So it appears that the double pulsing only affects xfer().

Regards,
Tony.

closed 2020-12-11 by TonyHSmith

semininja, 2020-12-11:
I'm not sure this is expected behavior, but at least it's a starting point for further investigation.