doceme/py-spidev #116: The CS_HIGH handling code triggers a lockup on Raspberry Pi in 5.10

pelwell, 2021-01-28:
SPI drivers that are configured to use GPIO descriptors, such as spi-bcm2835, always result in the SPI_CS_HIGH flag being set. This is meant to be purely internal state - the GPIO descriptor actually handles the polarity inversion as necessary - but the state is visible to clients of spidev.

This library intentionally issues zero-length reads if SPI_CS_HIGH is set, which is always true on Raspberry Pi kernels since 5.4. An optimisation present in the 5.10 and 5.11 kernels makes these reads fatal by triggering an almost-infinite loop. The lockup is the responsibility of the spi-bcm2835 driver, but you might want to reconsider whether this special SPI_CS_HIGH handling is still appropriate.

fgr1986, 2021-02-02:
@pelwell Indeed, i can confirm this causes an issue with latest kernels

fgr1986, 2021-02-02:
For those looking for a quick kernel rollback, reverting using https://github.com/Hexxeh/rpi-firmware/commits/master
**Note** that rpi3b was not working with kernel 5.4.83, so it was reverted to 5.4.79

* for **rpi3b+**: `sudo rpi-update  0642816ed05d31fb37fc8fbbba9e1774b475113f`
* for **rpi4b**: `sudo rpi-update 453e49bdd87325369b462b40e809d5f3187df21d`

HowardALandman, 2021-11-04:
There are two bugs here.
1. _spi.cshigh appears True when it actually should be False (SPI standard) by default.
2. Attempting to set it False crashes the program.
So the entirely reasonable-looking code:
`if self._spi.cshigh:
    self._spi.cshigh = False`
hits both bugs and dies.

closed 2022-11-03 by doceme