doceme/py-spidev #117: [RPi 4] Spurious CS pulse after transfer

eelcogg, 2021-01-29:
Hi,

I'm using spidev on a raspberry pi 4, python version 3.7.3. I am trying to communicate with an ADNS3080 optical flow sensor.
I am using CE0 (gpio8), MOSI (gpio10), MISO (gpio9), and SCLK (gpio11).
After each transfer on the SPI bus, the chip select line gives a spurious pulse 50us after going high at the end of the transfer. It stays low for roughly 11us. After 10us, MOSI also low for 1us until both go high again. This screws up the timing on the ADNS. I am sending a command to prepare it for clocking out data, but the extra clock pulse arrives before the ADNS is ready to clock out the data (it needs CS high for 75us to do a read)

My issue is similar to [this one](https://www.raspberrypi.org/forums/viewtopic.php?t=294385) and [this one](https://www.raspberrypi.org/forums/viewtopic.php?t=73323). Therefor I believe it is unrelated to the slave device I'm using. 

I have attached a logic analyzer trace (pulseview, saleae logic analyzer)
[image]

From the links above, this issue has existed since at least 2014. Right now I am bit-banging the chip select on another gpio as a workaround but have yet to test all scenarios.

I would be very grateful if you could look into this.

Thanks
[image]

MetaStasisa, 2021-02-12:
Hey @eelcogg 

See: https://github.com/doceme/py-spidev/issues/113

This is a duplicate issue - Hopefully the solution in that issue helps.

thomasfla, 2022-02-19:
Hi, Thanks for the nice lib!

I'm facing the same issue. 
[image]
Is there a roadmap to address this ? 
If lines :
https://github.com/doceme/py-spidev/blob/a5d82b88bd1c95c9f9ec46e9f7afe08c3323e625/spidev_module.c#L610 
https://github.com/doceme/py-spidev/blob/a5d82b88bd1c95c9f9ec46e9f7afe08c3323e625/spidev_module.c#L729
https://github.com/doceme/py-spidev/blob/a5d82b88bd1c95c9f9ec46e9f7afe08c3323e625/spidev_module.c#L882
are mandatory for some platforms only, could we make them optional without having to recompile, by adding a parameter, or a new version name?

closed 2022-11-03 by doceme