doceme/py-spidev #120: "xfer2"/"xfer" command erases slave data / slave is unable to respond

paulacvalle, 2021-04-08:
Hello everyone! 

I am working with a Raspberry Pi 3 B+ and Python 2.7, communicating with an LS7366R quadrature counter IC. 

This application really bugged me for days because I could send data to the boards set-up registers (MDR0/MDR1) and load data (DTR->CNTR). But every time I tried reading the counter register (CNTR/OTR), I could **read it only once before all reply bytes went to zero.**  I could read the other registers, but not this one. 

I tried running a similar code on Arduino and it works perfectly! So my guess is that there is an issue with my program/library:

1)   Either it forces the communication too much and causes the register to reset (unlikely, since it was happening even with low frequency)
2) There is an issue with the communication function, either the clock or chip select is not working properly, causing it to read/reply out of sync

Other than that, I couldn't think about any alternatives. I have no idea why I can read some registers, but not the counter ones (more than one single time).  I tried playing with frequency, "xfer2", "xfer", "writebytes", etc, but no luck.  

I gave up on using this library for now, but if anyone has run into a similar problem, let me know!

RobsonRojas, 2021-06-30:
@paulacvalle Did you find the cause of this? Experiencing the same problem.

paulacvalle, 2021-07-01:
> @paulacvalle Did you find the cause of this? Experiencing the same problem.

Unfortunately, not at all.  My solution was to use a different SPI communication library. 

This one is working perfectly for me: https://github.com/lthiery/SPI-Py 

Hope you find a solution!

aantoni4, 2021-07-23:
@paulacvalle Is there a way for you to post your solution to this problem?

paulacvalle, 2021-07-27:
> @paulacvalle Is there a way for you to post your solution to this problem?

Do you mean to post a tutorial on the other library? ahahaha

I didn't try to fix this issue with the spidev library, sorry. But I can make a tutorial on the other if it helps!

antgraf, 2022-08-16:
Probably it is caused by #117 #113

doceme, 2022-11-03:
Please retry with version 3.6

closed 2022-11-03 by doceme