doceme/py-spidev #125: Installation Python Spidev fails on Alpine Linux aarch64

cmonty14, 2022-04-17:
I'm trying to install Python Spidev on a RPi 4 Alpine Linux Edge OS.

Using this command `pip install spidev` for Python Spidev installation fails with this error:
```
Defaulting to user installation because normal site-packages is not
writeable
Collecting spidev     
  Using cached spidev-3.5.tar.gz (10 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: spidev
  Building wheel for spidev (setup.py) ... error
  error:
subprocess-exited-with-error                                           
                                                                                                                                                              
 
  × python setup.py bdist_wheel did not run successfully.
  │ exit code:
1                                                                                                                                              
 
  ╰─> [14 lines of
output]                                                      
      /usr/lib/python3.10/site-packages/setuptools/dist.py:717:
UserWarning: Usage of dash-separated 'description-file' will not be
supported in future versions
. Please use the underscore name 'description_file' instead
       
warnings.warn(                                                          
      running bdist_wheel
      running build
      running
build_ext                                                                                                                                       
 
      building 'spidev' extension
      creating build
      creating build/temp.linux-aarch64-3.10        
      gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3
-Wall -g -g -g -DTHREAD_STACK_SIZE=0x100000 -fPIC
-I/usr/include/python3.10 -c spidev_module
.c -o build/temp.linux-aarch64-3.10/spidev_module.o
      spidev_module.c:33:10: fatal error: linux/spi/spidev.h: No such
file or directory
         33 | #include <linux/spi/spidev.h>
            |          ^~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
   
  note: This error originates from a subprocess, and is likely not a
problem with pip.
  ERROR: Failed building wheel for spidev
  Running setup.py clean for spidev
Failed to build spidev
Installing collected packages: spidev
  Running setup.py install for spidev ... error
  error: subprocess-exited-with-error
   
  × Running setup.py install for spidev did not run successfully.
  │ exit code: 1
  ╰─> [16 lines of output]
      /usr/lib/python3.10/site-packages/setuptools/dist.py:717:
UserWarning: Usage of dash-separated 'description-file' will not be
supported in future versions
. Please use the underscore name 'description_file' instead
        warnings.warn(
      running install
     
/usr/lib/python3.10/site-packages/setuptools/command/install.py:34:
SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build
and pip and ot
her standards-based tools.
        warnings.warn(
      running build
      running build_ext
      building 'spidev' extension
      creating build
      creating build/temp.linux-aarch64-3.10
      gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3
-Wall -g -g -g -DTHREAD_STACK_SIZE=0x100000 -fPIC
-I/usr/include/python3.10 -c spidev_module
.c -o build/temp.linux-aarch64-3.10/spidev_module.o
      spidev_module.c:33:10: fatal error: linux/spi/spidev.h: No such
file or directory
         33 | #include <linux/spi/spidev.h>
            |          ^~~~~~~~~~~~~~~~~~~~
      compilation terminated.
      error: command '/usr/bin/gcc' failed with exit code 1
      [end of output]
   
  note: This error originates from a subprocess, and is likely not a
problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> spidev

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

```
Can you please advise how to fix this issue?

THX
Thomas

tim-seoss, 2023-03-09:
You must have the relevant Linux kernel headers installed on your system at build time.

https://pkgs.alpinelinux.org/contents?file=spidev.h&path=&name=&branch=edge

linuxtim, 2023-04-29:
@BANDWBEAR In general, you should open a separate issue instead of tagging on to an unrelated one.  I don't know much about MacOS but I assume that it doesn't have an spidev subsystem implemented (in the kernel or user space) at all (in which case py-spidev can never work at all).  If in-fact it does have an equivalent to Linux's spidev, then possibly you need to install the correct headers, or submit a PR to locate them correctly on MacOS.

closed 2025-09-25 by fschrempf