doceme/py-spidev #128: Supporting with-statements

erikbgithub, 2022-09-14:
As the library already has a class, an `open()` method and a `close()` method, it would make sense to add the small steps needed to make it callable in a `with` statement. However, as it's a c-Extension I currently don't feel confident that I could write the code alone.

In Python I would add this:

    class SpiDev(...):
        def __init__(self, param1, ...):
            ...
            self.param1 = param1
        ...
        def __enter__(self):
            self.open(self.param1)
    
        def __exit__(self):
            self.close()
    
        def __del__(self):
            self.close()

When the `with` statement is entered, the connection is opened. When it is left, the connection gets closed. Even in case of an exception. And just in case, the connection gets closed on object deletion as well.

More explanation regarding the [value of supporting `with`](https://stackoverflow.com/a/1984346).

Is there someone who has experience in transferring that idea to the C codebase here?

PS: Locally, I might simply add a wrapper class that gets the new methods added.

yochem, 2024-11-18:
It seems like this is currently implemented: https://github.com/doceme/py-spidev/blob/ac2671eba97f74a485154615ff740aaca2204bfe/spidev_module.c#L1409-L1432

This should support something like:

```python
with spidev.SpiDev() as spi:
   ...
```

However, with my level of understanding Python C modules, it seems like you would still need to call the `open()` method in the with-statement, and only the `close()` method is called for you:

How I think it would currently work:
```python
with spidev.SpiDev() as spi:
   spi.open(0, 0)
   spi.max_speed_hz = 4_000_000
   spi.writebyte([0x00])
```
I am currently also not able to test this.

How I wish it would work:
```python
with spidev.SpiDev(bus=0, device=0, max_speed_hz=4_000_000) as spi:
   spi.writebyte([0x00])
```

Or maybe even better, keeping the SPI object initialized but not opened:

```python
spi_device = spidev.SpiDev(bus=0, device=0, max_speed_hz=4_000_000)

with spi_device.open() as conn:
   conn.writebyte([0x00])

# do some other stuff here...

with spi_device.open() as conn:
   conn.writebyte([0xFF])
```

@doceme In general, would it be an idea to document the interface for this package? Maybe autogenerate API documentation? I also feel the current readme is not sufficient, as it for example lacks documentation about this feature. If you would be interested, I could investigate the possibilities.

fschrempf, 2025-09-25:
@yochem If you are still interested, feel free to go ahead and suggest changes to the documentation or the code. There is definitely some room for improvements.

fschrempf, 2025-09-25:
@yochem Thanks! Yes, I think I could do some testing and reviewing.

fschrempf, 2025-09-27:
@yochem Personally, I got some experience with Python and with C, but almost none with Python C modules, which is why I find it difficult too, to tell what would be the best way. But at first glace it makes sense to separate the low-level C code and the higher-level Python wrapper. Type hints for example would be really nice to have in my opinion.

If @doceme or anyone else doesn't have any objections, I would assume it is worth looking into it.