doceme/py-spidev #131: Could not build wheels for spidev

beyond-space21, 2023-09-09:
While attempting to install spidev through ```pip install spidev``` the process returned with an error as below,

```
Failed to build spidev
ERROR: Could not build wheels for spidev, which is required to install pyproject.toml-based projects
```

Vishnu-Naik, 2023-09-13:
> While attempting to install spidev through `pip install spidev` the process returned with an error as below,
> 
> ```
> Failed to build spidev
> ERROR: Could not build wheels for spidev, which is required to install pyproject.toml-based projects
> ```

@beyond-space21 This could be a cascade error, which means this error is caused by another error that happened earlier. Could you try to read the error thrown just before this error in the traceback error stack? 

As this package embeds C code, pip tries to compile the C code before installing. Hence a C compiler is required for this purpose. Sometimes, this process could be causing the issue, so I suggest installing a C compiler like gcc.

henla464, 2023-09-22:
I am getting the same error message. I have gcc installed.

`$ sudo pip3 install spidev --pre
Collecting spidev
  Using cached spidev-3.6.tar.gz (11 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: spidev
  Building wheel for spidev (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for spidev (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [69 lines of output]
      /usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'description-file' will not be supported in future versions. Please use the underscore name 'description_file' instead
        warnings.warn(
      running bdist_wheel
      running build
      running build_ext
      building 'spidev' extension
      creating build
      creating build/temp.linux-armv7l-3.11
      arm-linux-gnueabihf-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.11 -c spidev_module.c -o build/temp.linux-armv7l-3.11/spidev_module.o
      creating build/lib.linux-armv7l-3.11
      arm-linux-gnueabihf-gcc -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -g -fwrapv -O2 build/temp.linux-armv7l-3.11/spidev_module.o -L/usr/lib/arm-linux-gnueabihf -o build/lib.linux-armv7l-3.11/spidev.cpython-311-arm-linux-gnueabihf.so
      installing to build/bdist.linux-armv7l/wheel
      running install
      running install_lib
      Traceback (most recent call last):
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
          main()
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 261, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 230, in build_wheel
          return self._build_with_temp_dir(['bdist_wheel'], '.whl',
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 215, in _build_with_temp_dir
          self.run_setup()
        File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 158, in run_setup
          exec(compile(code, __file__, 'exec'), locals())
        File "setup.py", line 26, in <module>
          setup(  name            = "spidev",
        File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 153, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 148, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 163, in run_commands
          dist.run_commands()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 967, in run_commands
          self.run_command(cmd)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 335, in run
          self.run_command('install')
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "/usr/lib/python3/dist-packages/setuptools/command/install.py", line 68, in run
          return orig.install.run(self)
                 ^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/command/install.py", line 622, in run
          self.run_command(cmd_name)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 313, in run_command
          self.distribution.run_command(command)
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/dist.py", line 985, in run_command
          cmd_obj.ensure_finalized()
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 107, in ensure_finalized
          self.finalize_options()
        File "/usr/lib/python3/dist-packages/setuptools/command/install_lib.py", line 17, in finalize_options
          self.set_undefined_options('install',('install_layout','install_layout'))
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 290, in set_undefined_options
          setattr(self, dst_option, getattr(src_cmd_obj, src_option))
                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/lib/python3/dist-packages/setuptools/_distutils/cmd.py", line 103, in __getattr__
          raise AttributeError(attr)
      AttributeError: install_layout. Did you mean: 'install_platlib'?
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for spidev
Failed to build spidev
ERROR: Could not build wheels for spidev, which is required to install pyproject.toml-based projects
`

chachi, 2025-10-03:
Setting `SETUPTOOLS_USE_DISTUTILS=stdlib` in the environment fixed this for me.