doceme/py-spidev #140: Making the SpiDev file descriptor accessible.

sbelbin, 2025-07-04:
We have automated script that executes various test cases and much of it is implemented in Python script which uses a`SpiDev` object to perform IOCTL with Raspberry Pi devices.

The automated script needs to run at 1024 Hz (eventually to 2048 Hz) and thus the critical portions are implemented in C and invoked via Cython the bridging interface (adapter) between Python script and C.

Missing is capability of the `SpiDev` object to provide its underlying file descriptor (fd) that is missing.

_The C code won't compete with Python script for the file descriptor such as both using it at the same, nor does the C code close the the file descriptor, but leaves that to Python script via `SpiDev.close` or upon destruction of the `SpiDev` object._

Regards,
Steven

sbelbin, 2025-07-04:
That is correct.

I've got an idea what needs to be changed.

But, my policy is that it's best to ask. There might be reasons as not to provide the file descriptor property or even better someone has tried in the past then removed it.

fschrempf, 2025-09-25:
I guess if someone would like to implement this, it could be done. But on the other hand I'm wondering if it is really a good idea as there is no way to ensure that there is no interference. This would create an additional use-case that needs to be supported and maintained.

My suggestion would be that you use C code for the `SpiDev` handling (instead of `py-spdev`) altogether and only keep your test scripting as a Python wrapper.

yochem, 2025-09-29:
Isn't this what `SpiDev_fileno` does?

https://github.com/doceme/py-spidev/blob/60375bd992a6dae570593051bf802d4f8bb0264c/spidev_module.c#L913-L923

yochem, 2025-10-01:
Do I understand it correctly that you want it as a _property_ instead of a method? Because it's an available method:

https://github.com/doceme/py-spidev/blob/60375bd992a6dae570593051bf802d4f8bb0264c/spidev_module.c#L1469-L1470

yochem, 2025-10-03:
It's the standard way to return the underlying file descriptor in Python: https://docs.python.org/3/library/io.html#io.IOBase.fileno